esl/erlang_ale #16: Failed to compile erlang_ale

ethrbh, 2015-02-11:
hello,

Today I have started to learn and play with Erlang on Rasperry Pi B 1G RAM (http://www.raspberrypi.org/products/raspberry-pi-2-model-b/).

I have installed the latest erlang-mini, than I cloned the Erlang-ALE project from GitHub. I followed the instruction to compile the project. Unfortunately there are few errors when compile meck application what is one of the deps. project of Erlang-ALE.

I could fix it on my local copy, but for same reason I cannot able to push the changes in the git repo. See details below. I would be happy if someone could push the fix :-)

Log on the error
---------------------------------
pi@raspberrypi ~/ethrbh/temp/erlang-ale $ make all
make[1]: Entering directory '/home/pi/ethrbh/temp/erlang-ale/deps/gproc'
./rebar get-deps
==> gproc (get-deps)
./rebar compile
==> gproc (compile)
make[1]: Leaving directory '/home/pi/ethrbh/temp/erlang-ale/deps/gproc'
make[1]: Entering directory '/home/pi/ethrbh/temp/erlang-ale/deps/meck'
/bin/sh: 1: ./rebar: not found
/bin/sh: 1: compile: not found
Makefile:6: recipe for target 'compile' failed
make[1]: *** [compile] Error 127
make[1]: Leaving directory '/home/pi/ethrbh/temp/erlang-ale/deps/meck'
make[1]: Entering directory '/home/pi/ethrbh/temp/erlang-ale/deps/pihwm'

OS on raspberry pi:
----------------------------------
pi@raspberrypi ~/ethrbh/temp/erlang-ale $ uname -a
Linux raspberrypi 3.18.5-v7+ #225 SMP PREEMPT Fri Jan 30 18:53:55 GMT 2015 armv7l GNU/Linux

Solution what is working (for me at least)
----------------------------------------------------
Use the master branch of meck when clone in the erlang_ale/Makefile
old line:
     dep_meck = https://github.com/eproxus/meck.git 0.7.2
new line:
     dep_meck = https://github.com/eproxus/meck.git master

thanks for your help,
/Robi

knewter, 2015-02-11:
Robi,

Thanks for the heads-up.  My 2nd gen Raspi B is coming (presently out of stock at the supplier so backordered).  Once it arrives I'm trying to get a handle on this repo again :)  I'll try to get this change merged in then.

ethrbh, 2015-02-11:
hello Josh,

Very welcome :-) Right now I could step forward but another fault come :-( Compile the erlang_portutil is failing. I try to digin to it. 

I really would like to use this erlang extension on my Pi. So please let me know if you have done some improvements.

The new fault:
-------------------------------
Compiled src/meck_expect.erl
Compiled src/meck_cover.erl
Compiled src/meck_args_matcher.erl
Compiled src/meck.erl
Compiled src/meck_proc.erl
make[2]: Leaving directory '/home/pi/ethrbh/temp/erlang.mk/deps/meck'
make[2]: Entering directory '/home/pi/ethrbh/temp/erlang.mk/deps/pihwm'
make[2]: *** No targets specified and no makefile found.  Stop.
make[2]: Leaving directory '/home/pi/ethrbh/temp/erlang.mk/deps/pihwm'
make[2]: Entering directory '/home/pi/ethrbh/temp/erlang.mk/deps/erlang_portutil'
gcc -c -o portutil.o portutil.c -Wall -std=c99 -I/usr/lib/erlang/usr/include
make[2]: Leaving directory '/home/pi/ethrbh/temp/erlang.mk/deps/erlang_portutil'
mkdir -p priv ebin
gcc -I/usr/local/include -I/usr/lib/erlang/lib/erl_interface-3.7.17/include -I/usr/lib/erlang/usr/include/ -Ideps/erlang_portutil -Ideps/pihwm/lib -c -o priv/portutil.o deps/erlang_portutil/portutil.c 
gcc: error: deps/erlang_portutil/portutil.c: No such file or directory
gcc: fatal error: no input files
compilation terminated.
Makefile:37: recipe for target 'portutil' failed
make[1]: *** [portutil] Error 4
make[1]: Leaving directory '/home/pi/ethrbh/temp/erlang.mk/deps/erlang_ale'
erlang.mk:141: recipe for target 'deps' failed
make: *** [deps] Error 2
pi@raspberrypi ~/ethrbh/temp/erlang.mk $ 

thanks,
/Robi

ethrbh, 2015-02-11:
hello Josh,

I had to install autoconf tool to the Pi:
	pi@raspberrypi ~/ethrbh/temp/erlang.mk/deps/pihwm $ config_data ,../cls -la
	total 44
	drwxr-xr-x 6 pi pi 4096 Feb 11 19:55 .
	drwxr-xr-x 8 pi pi 4096 Feb 11 19:56 ..
	-rwxr-xr-x 1 pi pi 2298 Feb 11 19:55 autogen.sh
	-rw-r--r-- 1 pi pi 3658 Feb 11 19:55 configure.ac
	drwxr-xr-x 2 pi pi 4096 Feb 11 19:55 demo
	drwxr-xr-x 2 pi pi 4096 Feb 11 19:55 doc
	drwxr-xr-x 8 pi pi 4096 Feb 11 19:55 .git
	-rw-r--r-- 1 pi pi  408 Feb 11 19:55 .gitignore
	drwxr-xr-x 2 pi pi 4096 Feb 11 19:55 lib
	-rw-r--r-- 1 pi pi 1011 Feb 11 19:55 Makefile.am
	-rw-r--r-- 1 pi pi  855 Feb 11 19:55 README.md
	pi@raspberrypi ~/ethrbh/temp/erlang.mk/deps/pihwm $ ./autogen.sh 

	You must have autoconf installed to compile pihwm.
	Download the appropriate package for your distribution,
	or get the source tarball at ftp://ftp.gnu.org/pub/gnu/

	You must have automake installed to compile pihwm.
	Get ftp://sourceware.cygnus.com/pub/automake/automake-1.4.tar.gz
	(or a newer version if it is available)
	pi@raspberrypi ~/ethrbh/temp/erlang.mk/deps/pihwm $ apt-sudo apt-get install autoconf

but this does not solved all the issues. There is one left (or maybe more):
	Compiled src/meck_proc.erl
	make[1]: Leaving directory '/home/pi/ethrbh/temp/erlang_ale_ethrbh/erlang_ale/deps/meck'
	make[1]: Entering directory '/home/pi/ethrbh/temp/erlang_ale_ethrbh/erlang_ale/deps/pihwm'
	make[1]: *** No targets specified and no makefile found.  Stop.
	make[1]: Leaving directory '/home/pi/ethrbh/temp/erlang_ale_ethrbh/erlang_ale/deps/pihwm'

It looks pihwm project uses automake, and maybe this does not looks good for the make system.
	pi@raspberrypi ~/ethrbh/temp/erlang_ale_ethrbh/erlang_ale $ ls deps/gproc/pihwm/
	autogen.sh  configure.ac  demo  doc  lib  Makefile.am  README.md
	pi@raspberrypi ~/ethrbh/temp/erlang_ale_ethrbh/erlang_ale $ 
	pi@raspberrypi ~/ethrbh/temp/erlang_ale_ethrbh/erlang_ale $ 

I tried install the newest automake, but I have the latest:
	pi@raspberrypi ~/ethrbh/temp/erlang_ale_ethrbh/erlang_ale $ sudo apt-get install automake 
	automake is already the newest version.
	automake set to manually installed.
	0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
	pi@raspberrypi ~/ethrbh/temp/erlang_ale_ethrbh/erlang_ale $ 

Right now I don't know what should be the next step. I hope you can turn on the light in my head :-)

thanks for your help,
/Robi

ethrbh, 2015-02-12:
hello,

Just FYI, I raised a new issue on pihwm tool, due to failed compilation
    https://github.com/omerk/pihwm/issues/5

br,
/Robi

rclements, 2015-02-12:
Just tried building it on a first version RasPi and had no problems.  Seems to be a version 2 issue.  Knewter and I both are waiting to receive our RasPi2 boards so we can look into this further.

ethrbh, 2015-02-12:
hello rclements,

Thanks for your feedback.

I have read few documents about the new Pi HW and it has different CPU, more RAM, and more GPIO, than the 1st gen. HW. So the 1st 26 PIN on the 1st rev and 2nd rev HW is the same, and I though that if I use eq. GPIO-14 it should work. But If I see the error I got, it looks something is missing in my OS, so this is not that point that the GPIO access is wrong or not, I guess. 

I would like to ask you which OS do you use on your Pi? Maybe I can try use that, and see how it work for me.

I use this OS:
pi@raspberrypi ~/ethrbh/temp/erlang-ale $ uname -a
Linux raspberrypi 3.18.5-v7+ #225 SMP PREEMPT Fri Jan 30 18:53:55 GMT 2015 armv7l GNU/Linux

many thanks for your help,
/Robi

rclements, 2015-02-12:
I'm running Raspbian Linux 
Linux raspberrypi 3.10.25+ #622 PREEMPT Fri Jan 3 18:41:00 GMT 2014 armv6l GNU/Linux

ethrbh, 2015-02-12:
hello Robby,

Thanks, I try download and burn into the Pi, and see how it works.

thanks,
/Robi

rclements, 2015-02-12:
Let me know how it goes!

ethrbh, 2015-02-12:
hello Robby,

I have bad news :-( I could download the http://downloads.raspberrypi.org/raspbian/images/raspbian-2014-01-09/ and http://downloads.raspberrypi.org/raspbian/images/raspbian-2014-06-22/ images but Pi wont boot with these. I have to restore my original image what is running, but the ALE wont compile.
I think I need wait for your fix, but tomorrow I try digin what could cause that I cannot able to compile it. 

thanks for your help,
/Robi

knewter, 2015-02-12:
Here's another fun idea.  Perhaps we could set up a pair programming session on the pi.  Robby and I use tmate.io for this regularly, though we typically run it on a machine on his network then ssh into the pi together.  At any rate, if we could do that then at least we could pair with you in the environment (perhaps in conjunction with a Google Hangout for voice and video) and see if we could figure out the problem for you.  I'd be willing to try this tomorrow if you want.  You can email me at josh dot rubyist at gmail dot com.  Otherwise we'll have to wait until the pi2 arrives, and I don't even have an estimate on that date from the supplier yet sadly.

ethrbh, 2015-02-13:
hello Josh,

I have sent an email to you from my gmail address. Unfortunately I forgot fill the subject :-( The email has detailed info about how you can login to my Pi.

thanks for your help,
/Robi

knewter, 2015-02-13:
I just ssh'ed into your pi, followed the README, and erlang_ale compiled fine.  This is inevitably some issue with not using raspbian I guess?  Anyway let me know if that sorts you out?

ethrbh, 2015-02-13:
hello Josh,

Thanks that you have logged in to the Pi and compiled the code. I also tried once more from the beginning. Unfortunately the meck and pihwm deps are not compiled.

Compiled src/gproc.erl
make[1]: Leaving directory '/home/pi/ethrbh/erlang-ale-3/erlang-ale/deps/gproc'
make[1]: Entering directory '/home/pi/ethrbh/erlang-ale-3/erlang-ale/deps/meck'
/bin/sh: 1: ./rebar: not found
/bin/sh: 1: compile: not found
Makefile:6: recipe for target 'compile' failed
make[1]: *** [compile] Error 127
make[1]: Leaving directory '/home/pi/ethrbh/erlang-ale-3/erlang-ale/deps/meck'
make[1]: Entering directory '/home/pi/ethrbh/erlang-ale-3/erlang-ale/deps/pihwm'
make[1]: *** No targets specified and no makefile found.  Stop.
make[1]: Leaving directory '/home/pi/ethrbh/erlang-ale-3/erlang-ale/deps/pihwm'
make[1]: Entering directory '/home/pi/ethrbh/erlang-ale-3/erlang-ale/deps/erlang_portutil'
gcc -c -o portutil.o portutil.c -Wall -std=c99 -I/usr/lib/erlang/usr/include
make[1]: Leaving directory '/home/pi/ethrbh/erlang-ale-3/erlang-ale/deps/erlang_portutil'

I would like to ask you did I something wrong, or do I missed something?

thanks for your  help,
/Robi

ps: below you can see the whole logs about the compilation.

All logs from the compilation
--------------------------------------

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2015.02.13 19:41:17 =~=~=~=~=~=~=~=~=~=~=~=
login as: pi
pi@81.182.154.107's password: 
Linux raspberrypi 3.18.5-v7+ #225 SMP PREEMPT Fri Jan 30 18:53:55 GMT 2015 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Feb 13 15:59:04 2015 from rrcs-24-73-187-202.se.biz.rr.com

pi@raspberrypi ~ $ pwd
/home/pi
pi@raspberrypi ~ $ cd ethrbh/
pi@raspberrypi ~/ethrbh $ kmkdir erlang-ale-3
pi@raspberrypi ~/ethrbh $ mkdir erlang-ale-3 erlang-ale-3c erlang-ale-3d erlang-ale-3
pi@raspberrypi ~/ethrbh/erlang-ale-3 $ git clone https://github.com/esl/erlang-ale.git
Cloning into 'erlang-ale'...
remote: Counting objects: 867, done.
remote: Compressing objects:  11% (1/9)   
remote: Compressing objects:  22% (2/9)   
remote: Compressing objects:  33% (3/9)   
remote: Compressing objects:  44% (4/9)   
remote: Compressing objects:  55% (5/9)   
remote: Compressing objects:  66% (6/9)   
remote: Compressing objects:  77% (7/9)   
remote: Compressing objects:  88% (8/9)   
remote: Compressing objects: 100% (9/9)   
remote: Compressing objects: 100% (9/9), done.
Receiving objects:   0% (1/867)   
Receiving objects:   1% (9/867)   
Receiving objects:   2% (18/867)   
Receiving objects:   3% (27/867)   
Receiving objects:   4% (35/867)   
Receiving objects:   5% (44/867)   
Receiving objects:   6% (53/867)   
Receiving objects:   7% (61/867)   
Receiving objects:   8% (70/867)   
Receiving objects:   9% (79/867)   
[... 928 lines cut ...]
priv/portutil.o priv/pihwm.o priv/pi_gpio.o c_src/gpio_port.c \
-lpthread -lerl_interface -lei
erlc -o ./ebin src/gpio.erl 
# we need to re-compile bits of pihwm here because NIFs require -fPIC
gcc -Lpriv -L/usr/lib/erlang/usr/lib -L/usr/lib/erlang/lib/erl_interface-3.7.17/lib -I/usr/local/include -I/usr/lib/erlang/lib/erl_interface-3.7.17/include -I/usr/lib/erlang/usr/include/ -Ideps/erlang_portutil -Ideps/pihwm/lib -o priv/pwm_nif.so -fPIC -shared \
deps/pihwm/lib/pihwm.c deps/pihwm/lib/pi_gpio.c c_src/pwm_nif.c \
-lpthread
erlc -o ./ebin src/pwm.erl 
src/pwm.erl:63: Warning: variable 'Devname' is unused
gcc -Lpriv -L/usr/lib/erlang/usr/lib -L/usr/lib/erlang/lib/erl_interface-3.7.17/lib -I/usr/local/include -I/usr/lib/erlang/lib/erl_interface-3.7.17/include -I/usr/lib/erlang/usr/include/ -Ideps/erlang_portutil -Ideps/pihwm/lib -o priv/i2c_lib \
priv/portutil.o priv/pihwm.o priv/pi_i2c.o c_src/i2c_ei.c \
-lpthread -lerl_interface -lei
erlc -o ./ebin src/i2c.erl
gcc -Lpriv -L/usr/lib/erlang/usr/lib -L/usr/lib/erlang/lib/erl_interface-3.7.17/lib -I/usr/local/include -I/usr/lib/erlang/lib/erl_interface-3.7.17/include -I/usr/lib/erlang/usr/include/ -Ideps/erlang_portutil -Ideps/pihwm/lib -o priv/spi_lib \
priv/portutil.o priv/pihwm.o priv/pi_spi.o c_src/spi_ei.c \
-lpthread -lerl_interface -lei
erlc -o ./ebin src/spi.erl 
erlc -o examples examples/*.erl
pi@raspberrypi ~/ethrbh/erlang-ale-3/erlang-ale $ ls
c_src  deps  doc  ebin  erlang.mk  examples  Makefile  priv  README.md  src  test
pi@raspberrypi ~/ethrbh/erlang-ale-3/erlang-ale $ ls deps/
erlang_portutil  gproc  meck  pihwm
pi@raspberrypi ~/ethrbh/erlang-ale-3/erlang-ale $ ls deps/pihwm/
autogen.sh  configure.ac  demo  doc  lib  Makefile.am  README.md
pi@raspberrypi ~/ethrbh/erlang-ale-3/erlang-ale $

ethrbh, 2015-02-13:
hello Josh,

I have a good news. I tried to make the erlang-ale once again, I got the same faults what I got before, but, I tried start an erlang shell (make shell) and just turn ON/OFF GPIO17. And it is working, the LED I plugged on that pin started to work :-) Now I'm interesting about meck and pihwm compilation error. Are these critical or ?

many thanks for your grate support and helping me.
/Robi