esl/erlang_ale #32: Just cant make it work

G3UKB, 2018-05-27:
Using PCA9685 i2c board with 2 servos attached. This works fine in other languages so no hardware issue. It seems that only the low 5 bits are set in the registers. Simple test.
test() ->
	{ok,Device} = connect(),
	i2c:write(Device, <<16#01, 16#FF>>),
	i2c:write(Device,<<16#00>>),
	io:format("Registers: ~w~n",[i2c:read(Device,20)]).
Output is:
Registers: <<31>>.
I did a full implementation of the device before I discovered nothing worked.

fhunleth, 2018-05-27:
The first `i2c:write` writes 0xff to register 1. The datasheet says that the top 3 bits of register 1 (MODE2 - see table 6) always read as 0, so I'd expect you to get back 31 if you read register 1 back. The next `i2c:write` and `i2c:read` look like they should read back the first 20 registers on the device instead of the just one that is printed. I'm guessing that you didn't paste the results of that run. However, looking at Table 7 in the datasheet, it looks like most (all?) of the registers force the top 3 bits to 0. That would be consistent with what you're seeing and would seem like the expected behavior.

fhunleth, 2018-05-27:
Here's the datasheet that I was looking at: https://www.nxp.com/docs/en/data-sheet/PCA9685.pdf

G3UKB, 2018-05-27:
Thanks for looking into it. I would really like to get this working.
[device.txt](https://github.com/esl/erlang_ale/files/2042984/device.txt)
Ok, so not great with erlang yet. Here is the complete file that I think should be a copy of this one that is the Adafruit python implementation. The python one works but the erlang one makes no attempt to move the servos.
[PCA9685.txt](https://github.com/esl/erlang_ale/files/2042990/PCA9685.txt)

G3UKB, 2018-05-28:
Figured it out by printing out all the registers in a working version and then in the erlang version. Mistake was in the set_pwm that was corrupting MODE1 because evaluation in wrong order, also wrong defines from copy and paste.

fhunleth, 2018-05-28:
Glad it works!

closed 2018-05-28 by fhunleth